just-trey/Marlin #2: Neptune 2D config changes.

mlee12382, 2022-01-05:
### Describe the feature you want

Below are the changes I have made to the config files that I believe will make it work on the Neptune 2D the lines with * I believe are all required and the others are features that I saw that looked like would be good to have enabled for all of the Neptune printers a couple of them are BLTouch related and some aren't.
Configuration.h Changes:
*	#define CUSTOM_MACHINE_NAME "Elegoo Neptune 2D"
*	#define EXTRUDERS 2
*	#define SINGLENOZZLE
*	#define E1_DRIVER_TYPE A4988
	#define Z_MIN_PROBE_REPEATABILITY_TEST
	#define PREHEAT_BEFORE_PROBING
*	#define NUM_RUNOUT_SENSORS   2
*	#define WATCH_ALL_RUNOUT_SENSORS
	#define PREHEAT_BEFORE_LEVELING
	#define G26_MESH_VALIDATION
	#define NOZZLE_CLEAN_FEATURE
	#define NOZZLE_CLEAN_MIN_TEMP 200
	
Configuration_adv.h Changes:
	#define FAN_KICKSTART_TIME 100
*	#define E1_AUTO_FAN_PIN FAN1_PIN
	#define ASSISTED_TRAMMING
	#define ASSISTED_TRAMMING_WIZARD
	#define BABYSTEP_DISPLAY_TOTAL
	#define SAVED_POSITIONS 1
*	#define TOOLCHANGE_FILAMENT_SWAP
*	#define TOOLCHANGE_FS_LENGTH              80
	
Version.h Changes:
*		#define MACHINE_NAME "Elegoo Neptune 2D"

mlee12382, 2022-01-05:
Edit v1:  PREHEAT_BEFORE_PROBING conflicts with PREHEAT_BEFORE_LEVELING so I left the former commented out. Also left tramming off as the probing locations had conflicts may revisit later if I decide I really want it enabled. Successfully compiled firmware following those changes and will test on printer and update progress.

mlee12382, 2022-01-07:
[Neptune2DConfigFiles.zip](https://github.com/just-trey/Marlin/files/7825528/Neptune2DConfigFiles.zip)
Here are the Config files, they have my personal pid values and offsets and e-steps so you will obviously want to change those back to "stock" I haven't tested Assisted tramming yet and I am having issues with getting the g26 mesh validation to print correctly, it is only printing in the corner of the bed and not moving for some reason. You can comment that back out though since it's not required just seemed like it would be a useful feature to have enabled. Mostly the changes needed are listed above in the body of the issue and #define DISTINCT_E_FACTORS

just-trey, 2022-01-07:
@mlee12382 thanks looking!

mlee12382, 2022-01-08:
[Neptune2DConfigFilesV2.zip](https://github.com/just-trey/Marlin/files/7832491/Neptune2DConfigFilesV2.zip)
Here is the corrected files, I fixed a couple of your typos in comments and redefined it as a single nozzle and removed the extra temp sensor issue that was causing my crash and a couple other minor issues. 
For distribution you will need to reset the xyz offset for the probe to the default elegoo spacing and also change the tramming location to match, maybe set up a definition and use variables so that anyone that needs to change it and compile they just change the variable value, like so:
#define PROBE_X 32.55 
#define PROBE_Y -7 
#define PROBE_Z 0 
#define NOZZLE_TO_PROBE_OFFSET { PROBE_X, PROBE_Y, PROBE_Z }
#define TRAMMING_POINT_XY { {  PROBE_X, 35 }, { 200,  35 }, { 200, 200 }, { PROBE_X, 200 } }

just-trey, 2022-01-09:
@mlee12382 I really appreciate all your help in making sure this works for everyone!

closed 2022-01-09 by just-trey