just-trey/Marlin #4: [BUG] Rare random blobs on prints printing over octopi on an PI Zero 2 W

escapppe, 2022-01-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Rare random blobs on prints printing over octopi on an PI Zero 2 W.

Printing over Octopi with a Pi Zero 2 E leavse some rare random blobs on the print. It prints total fine over SD Card.

I suspect the baudrate to be to high as elegoo sets the baudrate to be 115200. But maybe there is some other host - serial connection - printer problem.

Dont have the problem on stock fw.

### Bug Timeline

first

### Expected behavior

print clean 

### Actual behavior

blobs on the print

### Steps to Reproduce

print a file

### Version of Marlin Firmware

2093

### Printer model

neptune 2

### Electronics

pi zero 2 W

### Add-ons

none

just-trey, 2022-01-09:
I am implementing a new release today. Please check if that resolves the issue once released. If it does not, my new release will allow the use of`  M575` to change the baud rate. So you may want to see if that resolved the issue. Please let me know once you tried it out.

just-trey, 2022-01-09:
@escapppe  new version is out. See if it does any better and you try changing the baud speed with https://marlinfw.org/docs/gcode/M575.html

escapppe, 2022-01-11:
@just-trey with the baudrate set to 115200 I have no random blobs anymore. 

I will check if I can reproduce the problem with the new firmware and the baudrate set to 250000.

just-trey, 2022-01-11:
@escapppe I found this interesting nugget out there in this thread. Thinkyhead is the lead of Marlin so we can take his word for it. https://github.com/MarlinFirmware/Marlin/issues/12174

[image]

 I will set back to default to 115200

just-trey, 2022-01-11:
Change checked in and will be included in the next release. Thanks @escapppe

commit 3e587b5a (2022-01-11): fixes #4
closed 2022-01-11 by just-trey