just-trey/Marlin #10: [Question] high pitch fan noise under 100%?

escapppe, 2022-01-12:
### Is your feature request related to a problem? Please describe.

right now im using two 5015 blower fans that had a really high pitch noise on the latest release when running under 100%. 
Is it the same for the stock 4010 fan?

Enabling FAN_SOFT_PWM in configuration.h with different settings on SOFT_PWM_SCALE and SOFT_PWM_DITHER did not work for me on my blower fans. 

What worked for me is enabling FAST_PWM_FAN in configuration_adv.h and setting FAST_PWM_FAN_FREQUENCY to 20. Now it's nearly gone and like stock elegoo firmware. 

### Describe the feature you want

1

escapppe, 2022-01-12:
I tested several frequency values and like 57 (19*3) most for my 5015 blower fans. They are not silent but during printing the low ticking noise is acceptable and way better than the high pitch noise I had without that setting.

just-trey, 2022-01-16:
@escapppe testing this out today

just-trey, 2022-01-17:
@escapppe I am going to leave fan PWN control off based on my research. Looking at all the sample configurations of marlin out there all but one has it turned off. It definitely can fix your sound issue but I am not sure of any ramifications it may have [based on the comments in the file](https://github.com/MarlinFirmware/Marlin/blob/2.0.9.3/Marlin/Configuration_adv.h#L566) and if you find it useful in your own configuration that's great!

closed 2022-01-17 by just-trey