just-trey/Marlin #9: [Question] Junction Deviation at 0.013?

escapppe, 2022-01-12:
### Is your feature request related to a problem? Please describe.

[Question]

On the current release "CLASSIC JERK" (the one you could set in cura) is deactivated and the newer function: "JUNCTION DEVIATION" is enabled with the value 0.013. 

Is there anything behind the value 0.013 made for the neptune 2 or is this the premade marlin value?

in my first research on this topic, it was recommended to switch to the classic JERK mode for printers with independent X and Y axes.

### Describe the feature you want

^

just-trey, 2022-01-12:
@escapppe Nice research. let me know if you think it needs to be changed. Those numbers and (the fact it is enabled) are the base Marlin defaults.

escapppe, 2022-01-12:
So I switched to classic jerk. The E-JERK value in configuration.h has to be set to at least 10 for a bowden setup. Otherwise the firmware won't compile with linear advance enabled. 

I had no problems with the classic jerk and prints came out as fine or maybe even better as with junction deviation. 

My jerk settings are set to 8 on Cura but I made no calibration.

just-trey, 2022-01-17:
I am moving back to classic jerk after further research on my side. Seem like the preferred method for Cartesian printers.

closed 2022-01-17 by just-trey