just-trey/Marlin #11: Neptune 2D V1.2 Touch screen calibration getting stuck

Igaveuponfixingithub, 2022-01-17:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Marlin is freezing up after I tap the 4 calibration points reinstalled multiple times and attempted to reboot multiple times

### Expected behavior

Marlin to run after caliration

### Actual behavior

Marlin freezes up after calibrating saying tap left again and getting stuck

### Steps to Reproduce

Use marlin V1.2 Neptune2D BLTouch 
Boot printer
Go thru calibration
bug happens after fourth point

### Version of Marlin Firmware

Neptune 2D V1.2 BLTouch

### Printer model

Neptune 2D

### Add-ons

BLTouch

### Additional information & file uploads

Used Neptune 2D-BLTOUCH v1.2

Igaveuponfixingithub, 2022-01-17:
Fix was removing SDcard during calibration and then reinserting when it said calibration complete

closed 2022-01-17 by Igaveuponfixingithub

ajshell1, 2022-01-23:
I also had this issue on the 2S with 1.3's BLTOUCH file, but I solved it in the same way as you. Thank you for providing this solution.