just-trey/Marlin #13: [FR] Changing driver to have full color spectrum on Elegoo Neptune 2/2S/D

escapppe, 2022-02-04:
### Is your feature request related to a problem? Please describe.

Changing the Driver in Marlin\src\inc\Conditionals_LCD.h 
on Line: 1265 from ILI9488 to ILI9341 adds full color options (white is white not light green)

### Describe the feature you want

Colors

just-trey, 2022-03-01:
Holy cow I just saw this. I'll look at it ASAP!

just-trey, 2022-03-22:
@escapppe testing a release hopefully will include this tomorrow

Babble69, 2022-03-22:
Hey, just-trey I am having a problem with my install.  I'm a total noob to the firmware side of things, but know everything about the printer side, can you please contact me back and I can explain what is going on? Thanks!!

just-trey, 2022-03-23:
@Babble69  I need a way to contact you. Email? Discord? Something else?

just-trey, 2022-03-24:
https://github.com/just-trey/Marlin/releases/tag/2.0.9.3-N2-rc3 resolved this. @escapppe  I was able to do this in  [configuration.h](https://github.com/just-trey/Marlin/blob/933c0fa99dd5d0e13d4c62106cae115a9b2f4f3a/config/Elegoo/Neptune-2/Configuration.h#L2839-L2853) rather than have to change the file you pointed to.

closed 2022-03-24 by just-trey