just-trey/Marlin #27: [BUG] stuck on 100% update

Duudeeee, 2023-01-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I just can use the old marlin version 2.0.9.3 
Every time i try to update to yours new version i got stuck at 100% update.
My board is 1.3 
I dont understend 

### Version of Marlin Firmware

2.0.9.3

### Printer model

Neptune 2s

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

SDLIEB, 2023-03-10:
I have this same problem on my 2D with a 1.3 board and now my machine is stuck at booting.

RonnyLV, 2023-03-12:
Re-formatting sd card to fat32 and trying again should help